danpeig/ESP32EnergyMonitor #2: It will not read the third phase. Real Power and Power Factor

vladakru, 2023-03-19:
Greetings, I am very pleased with the project you made.
I have a problem with reading Power Factor P3 and Real Power P3 is always 0.00, P1 and P2 print values ​​normally. 
I must mention that Apparent Power P3 and Current P3 also print values ​​normally. 
I am afraid that this affects the Total Real Power so that the Monthly and Daily Consumption are incorrect.
How could I solve this problem?
THX
[image]

danpeig, 2023-03-19:
Hi,

Did you tried inverting the direction of the current sensor for P3?

vladakru, 2023-03-19:
I didn't turn the sensor because during the calibration process it was in the same direction as it is now and it worked like that, so I suspected that it might not be a problem in the code...

vladakru, 2023-03-19:
how does the Curen P3 measure me and it won't count the Real Power P3...???

danpeig, 2023-03-19:
Can you run the calibration script again replacing this code block:
` 
  Serial.print("V3: ");
  Serial.print(supplyVoltage3);
  Serial.print(", I3: ");
  Serial.println(Irms3);
`

with this

` 
  Serial.print("V3: ");
  Serial.print(supplyVoltage3);
  Serial.print("PF3: ");
  Serial.print(powerFactor3);
  Serial.print(", I3: ");
  Serial.println(Irms3);
`

What we are looking for is a negative power factor.

vladakru, 2023-03-19:
with no load:
------------
V1: 229.00, I1: 0.92
V2: 225.42, I2: 0.88
V3: 226.79PF3: -0.02, I3: 1.02

with load:
V1: 221.39, I1: 13.11
V2: 218.49, I2: 13.28
V3: 220.35PF3: 0.85, I3: 12.92

danpeig, 2023-03-19:
The hardware appears to be working well. 3 positive parameters.
Now we must check ESPHome and Home Assistant.
When you run the ESPHome console/terminal, do you get similar results in the output variables (voltage, current, and PF)?

vladakru, 2023-03-19:
This is report from pc terminal directly connected to ESP32
[image]

danpeig, 2023-03-19:
Ok, the problem is not in Home Assistant.

Would you mind sharing your _esp32emon.yaml_ and _esp32emon.h_ sources? You can change any sensitive information. I just want to compile and test here.

vladakru, 2023-03-20:
esp32emon.yaml
`# Platform settings
esphome:
  name: esp32emon
  name_add_mac_suffix: true
  includes:
    - emonlib/EmonLib.h
    - emonlib/EmonLib.cpp
    - esp32emon.h
  project:
    name: danpeig.esp32emon
    version: "1.3.0"

# Board configuration
esp32:
  board: esp32doit-devkit-v1
  framework:
    type: arduino

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# Sets up the improv via serial client for Wi-Fi provisioning
improv_serial:

# Enable logging
logger:

# Enable Home Assistant API
api:

# Enable OTA updates
ota:
  password: !secret ota_password

# Wifi configuration
wifi:
  networks:
    # Main network
   - ssid: !secret wifi_ssid
     password: !secret wifi_password
    # Alternative network
    # - ssid: !secret wifi_ssid2
      # password: !secret wifi_password2

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  # ap:
    # ssid: "Esp32EnergyMeter"
    # password: !secret wifi_ap_password

captive_portal:

# MQTT Integration
#mqtt:
#  broker: !secret mqtt_broker
#  port: !secret mqtt_port
#  username: !secret mqtt_user
#  password: !secret mqtt_password
#  discovery: false #We are using the default API for Home Assistant discovery, no need for MQTT

# Timers configuration
time:
  - platform: sntp
    id: sntp_time1
    on_time:
      # Every Month at specified day
      - seconds: 0
        minutes: 0
[... 565 lines cut ...]
    float current_total = emon1.Irms;
    current_sensor_total->publish_state(current_total);
    */
	
	/*
    // Totals 2 phases - uncomment only this block if you are reading two phases
    float realPower_total = emon1.realPower + emon2.realPower;
    realpower_sensor_total->publish_state(realPower_total);
    float apparentPower_total = emon1.apparentPower + emon2.apparentPower;
    apparentpower_sensor_total->publish_state(apparentPower_total);
    float current_total = emon1.Irms + emon2.Irms;
    current_sensor_total->publish_state(current_total);
	*/
    
     // Totals 3 phases - uncomment only this block if you are reading three phases
     float realPower_total = emon1.realPower + emon2.realPower + emon3.realPower;
     realpower_sensor_total->publish_state(realPower_total);
     float apparentPower_total = emon1.apparentPower + emon2.apparentPower + emon3.apparentPower;
     apparentpower_sensor_total->publish_state(apparentPower_total);
     float current_total = emon1.Irms + emon2.Irms + emon3.Irms;
     current_sensor_total->publish_state(current_total);
     
  }
};
`

vladakru, 2023-03-20:
thanks for your time, I'll try flashing again and see if it works.
thanks again

vladakru, 2023-03-20:
As you saw on the test table, everything worked as it should, when I connected it to the mains, one phase read amperes again, but not PF and RP, when I turned the CT sensor in the opposite direction, that phase worked with the readings. I don't know how it's like the current is flowing in the opposite direction or whatever... I hope that now with that setup the reading will be fine.

danpeig, 2023-03-26:
Here is the explanation from OpenEnergyMonitor:

Does the direction the CT is clipped on the wire, matter?

If you are interested in reading only apparent power, and you don’t have an AC-AC adapter, it does not matter. The power reading will always be positive.

If you want to read real power, and you want to know in which direction power is flowing – e.g. when you generate your own electricity, and want to know whether you are importing or exporting – then the C.T. must face the correct direction. Our convention is: imported and generated powers are positive. If the C.T. faces the wrong way, it means that the power you expect to be positive will be shown as negative. [Note: Robin Emley has adopted the opposite convention for his Mk 2 energy diverter – he regards exported power as positive.]

For the UK ac-ac adapter and the YHDC C.T. purchased from the shop and when the C.T. is on the line conductor, the printing on the face of the C.T. should face in the direction of positive power, i.e. in the picture above, positive power flow is from left to right. If the C.T. is on the neutral conductor (this is only valid on a single-phase installation) it should face in the opposite direction. For any other combination of ac-ac adapter and C.T. it is possible, if the phasing of each is known, to determine the correct orientation, but the easiest and quickest method is probably by trial and error. Reversing the C.T. will change the sign of that input. Reversing the ac-ac adapter (where the plug design allows this) will change the sign of all inputs.

closed 2023-03-26 by danpeig