danpeig/ESP32EnergyMonitor #5: Is the offset of 2.5V not an issue

paulca99, 2023-04-13:
Hi Dan , Great readme, I'm about to build this, I've been running on an arduino mega for years. I was just wondering if you'd tried using 3.3V as the Vcc onto the zmpt101b to get the voltage centred correctly around 1.666V , or doesn't it really matter ? Seems a shame to lose such a big chunk of the ADC resolution.

danpeig, 2023-04-14:
During my research I found that the module would provide more consistent results if using 5V supply and found that it worked well. Never tried the 3.3V option.

https://surtrtech.com/2020/04/08/measure-any-ac-voltage-250vac-with-zmpt101b-and-esp8266-12e-with-android-app-adafruit-io-mqtt/

paulca99, 2023-04-14:
Interesting, he's using 3.3V in that example and it seems to work ok.  It's a bit confusing as the write up says "the signal will be lost at some points using 3.3V"  but I think he's just talking about the clipping , which can be cancelled out with the pot. I'll try both.

danpeig, 2023-04-14:
This user proposed an interesting solution but requires hardware modifications: https://github.com/a-boudersa/ZMPT101B-3.3V-ADC

closed 2023-12-24 by danpeig