danpeig/ESP32EnergyMonitor #10: I want to use ACS712

Dydipanshu, 2023-12-24:
Hey, First of all very great work .....thanks for this.

But I wanted to ask you if it is possible to make this using ACS712 current sensor. If yes,then can you please tell me what connections should i make as I am having a hard time in figuring it out as it has completely different pins and input then the one used by you.

danpeig, 2023-12-24:
Hi,

The ACS712 as far as I know is rated for 5V. If you can find one that works with 3V, you can connect VOut to the current sensing GPIO and use the code as it is.

closed 2023-12-24 by danpeig

Dydipanshu, 2023-12-26:
Is there a way I can use voltage divider circuit.....if yes can you please provide me with a cirucit diagram where everything is connected as your current circuit diagram is confusing to me.

Everything connected means
ESP32,ZMPT101B,ACS712,Voltage divider circuit,Supply,Load

I will be very thankful to you as I have been trying to find and do it for the past 15 days but nothing seems to be working.

danpeig, 2023-12-28:
Hi,

I don't have experience with the ACS712 neither have some of that modules laying around to play with.

My advice is use an oscilloscope to ensure your physical circuit is fine and only then move to the ESP. The ADC of ESP32 is not that simple to work with.