danpeig/ESP32EnergyMonitor #11: PJ-307 pinout

Caddyman68, 2024-01-22:
Remember - there are no stupid questions. Mine is "where do I connect a)GPIO35 and b)the convergence of the 2 main resistors+ the positive side of the capacitor?" I understand that GPIO35 connects to "tip" and the other side to "sleeve" and I know what that is (I think) on the SCT013, but which of the 5 tabs on the PJ-307 are they? The pinouts I've seen - Ali Express for example - number them 1 (looks like sleeve) then 5-4 (normally closed) and 3-2 (normally closed). With plug inserted, I would think tip would be 5 (or 2) since it would disconnect from 4 (or 3)
[image]

I have tried different combinations with no luck; possibly have messed everything up already. I tested the voltage divider circuit with a multimeter and it seems okay. I am only using the basic ESPHome configuration for CT clamp - I notice you referenced this in issue #1 "Using without voltage meter". I have the ZMPT101B but wanted to see if I can get the CT clamp to work first. Also, I have the 0-1V output sensor so I don't have the burden resistor in the schematic. I have tried each of the 2 SCT013s I got from Ali; I tried testing with a multimeter but the range is too high on my "good" one (20v) so I used an old mini voltmeter on 2 volts but I seem to get voltage between all 3 contacts (tip/ring/sleeve). I hope it's DC I'm supposed to be reading. I have the SCT013 clamped around a lead to my heater coil which a clamp-on ammeter reads 21 amps consistently on when energized; I get no change in the amperage reading in HomeAssistant on the sensor created by ESPHome whether energized or not.
Thanks in advance for your help and indulgence. I appreciate the time you took to share your project with the community

Rhys79, 2024-02-20:
Pin 5 is Tip, Pin 2 is Ring, Pin 1 is Sleeve (TRS = Tip Ring Sleeve).  Pin 3 & 4 are used by more advanced equipment to determine: A - if there is a plug currently inserted into the jack (continuity broken between 2/3 & 4/5) and B - if the inserted plug is a TS (Tip/Sleeve aka Mono), or TRS (Tip/Ring/Sleeve aka Stereo).  Add an addition contact/breaker pair and you get a TRRS (Tip/Ring1/Ring2/Sleeve) jack, typically used by headsets, the additional Ring terminal used for the mono microphone input with ground common between mic and speakers.

Caddyman68, 2024-02-22:
> [Rhys79](/Rhys79)

Thanks for the help. I took one of the jacks apart but it's molded plastic (and pretty tiny) so it fell apart mostly. I could tell the tip was pin 5 and ring (middle section) looked like it should be 2, but it didn't seem like the last section (sleeve) was making contact with pin 1. Again, by this time it had fallen apart. The use of only pin 1 and 5 makes sense of the photo in the article - I only see a wire and a resistor and it almost looked like they were wired at the same point. Of course now I see that the wire goes through the board and connects to pin 1 while it is more apparent that the resistor goes to pin 5 at the bottom which also answers my question (not asked) about whether or not the pinout drawing of the PJ-307 was from the top - which it is.
 I connected pin 5 to GPIO35 - didn't use a burden resistor since my SCT013 is the voltage variety. I connected pin 1 to the voltage divider (point where the 2 resistors meet the capacitor). I have yet to test - I repaired the wiring on the heater I used so need to get some ambition to "strip" it again.
I guess a big part of my confusion involved NOT connecting the ring. As a side story, I used to work on telephone lines where tip & ring were the norm and ground - basically sleeve - was bad LOL. Not sure I even connected pin 1 in any of my trials.
Thanks again for your detailed explanation. I could tell that inserting a jack would break continuity between 2 - 3 and 4 - 5, but couldn't comprehend how that would be useful (ie why you would even need 3 and 4). Hopefully I'll get ambitious tomorrow and test. I will wait a few days and update before closing the issue even though my question is answered now

Rhys79, 2024-05-10:
Sorry for the long delay, things have been "eventful" in my personal life as of late.

Glad I was able to help!  My father was an EE, and I've been in IT for decades and have been playing with IOT designs since before IOT was a thing, but never as more than a hobby.  I remember how difficult it was at first learning how to read datasheets and understand diagrams, so I'm always happy to answer a question here and there to help makes things easier for newcomers than I had it.  I'm glad you found my explanation both helpful and easy to understand.  Feel free to ping me if you ever run into a similar question and need help :)

Caddyman68, 2024-05-10:
Thanks again for the reply. To be honest, I completely forgot about the question and to close the issue. I tested and it didn't work - no doubt something I did wrong other than the jack (soldering maybe since the PJ-307 won't just sit on a breadboard). I had intended to just cut the cord and hard wire it, but in the interim I ordered a completed project (Energy meter) from Ali that actually came without the plug/jack combo. The stripped wires (red & black) wire to a Zigbee device that looks similar to a Sonoff mini which interfaced easily with Home Assistant. I lost interest or I'd have stripped the SCT013 and tried it with the new device just to see if it was defective (or I broke it). 
We are indeed luckier today than just a few years ago that there is a forum to share knowledge/information and people like you willing to help. Thanks again - this time I will close the issue.

closed 2024-05-10 by Caddyman68