danpeig/ESP32EnergyMonitor #12: Calibration of SCT013 current sensor

zamojski, 2024-03-08:
The calibration program is supposed to calculate `CI1` etc. values. But what are these factors in real world? Is it a power factor of calibrated load?

zamojski, 2024-03-22:
I've troubles adapting this code to work on Kincony M16v2 where I've fully calibrated SCT013 sensor but only apparent power calculations. On the below screenshot the voltage reading is accurate (~0.2V error). I've calibrated `CI1` but I don't know if this approach will work for for varying current (like standby using 5W and PF 90 and turned on 35W PF 45).

Nonetheless power factor doesn't seem to be right. @danpeig Can you please advise? 
[image]

zamojski, 2024-03-22:
Here on pure resistive load when turned on:
[image]
and when load is unplugged:
[image]

danpeig, 2024-11-20:
What is the range of your SCT probe? 33W is too low, probably below the sensor range.

zamojski, 2024-11-20:
10A - the smallest I could buy.