danpeig/ESP32EnergyMonitor #13: Real power  phase 3

iankds05, 2024-11-20:
[image]

I wonder if you could shed any light as to why the only value I don't seem to be getting is the Real power calculation on P3
My set up has all the settings arranged to monitor 3 phases as per your write up and everything seems to conform to the correct values after calibration of voltages and actual current measured with calibrated digital meter.
I have gone through the code files both for emonlib and the yaml but nothing jumps out as being corrupted or incorrect formation and compilation in ESPHome works fine. I have swapped out the CT's and tested all connections but still no output for P3 kW. I only use one voltage measurement  input for all 3 phases. Any suggestions would be appreciated

iankds05, 2024-11-20:
Yep that's because p3 is dedicated to the thermal water heater which is controlled by thermostat. When its on, gives a current reading of 7.3A but real power is still not calculated. I'll try to capture a photo grab when its on to show

danpeig, 2024-11-20:
Did you tried to invert the current sensor in the wire?

iankds05, 2024-11-20:
Yes and swap it out for a brand new one as well

iankds05, 2024-11-20:
[image]

This is the issue with current and voltage being obtained but no Real Power Calculation

iankds05, 2024-11-20:
[image]

Everything else works just perfect apart from the calculations on P3 phase

danpeig, 2024-11-20:
What output do you get from CalculeteVI sketch for sensor 3?

iankds05, 2024-11-20:
When I did the calibration before installing I used a room heater drawing 1.5A for all three CT's and they were all withing 1% tolerance using the following cal settings. CT's are 100A 50ma open clamp type
#define CV1 470
#define CV2 470
#define CV3 470

#define CI1 41
#define CI2 42
#define CI3 42

The real current readings for P1 and P2 are acceptable tolerance measured with DM

iankds05, 2024-11-20:
Ok thanks for that info  I'll check that and get back to you.

iankds05, 2024-11-21:
[image]

So took you advice and added the "-" to code and it appears to have worked. No idea what the anomaly is with that phase but for now it working fine. Thanks for you input, I doubt I'd have figured out the fix

danpeig, 2024-11-21:
Great! When I have time I will review the code to use the absolute values instead of zero outputs. The power factor is correct? It should be close to 1 for resistive loads.

closed 2024-11-22 by danpeig

marcinnap76, 2024-11-22:
> ![Murray_Energy](https://private-user-images.githubusercontent.com/11013066/388731023-96cd3a2e-be6e-4a8a-af92-31acc616518a.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzIyNTM2NzksIm5iZiI6MTczMjI1MzM3OSwicGF0aCI6Ii8xMTAxMzA2Ni8zODg3MzEwMjMtOTZjZDNhMmUtYmU2ZS00YThhLWFmOTItMzFhY2M2MTY1MThhLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTIyVDA1MjkzOVomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTRmNTlkZDNiYTI3NWVhMTZhOTJkNDM2ODQ3MTY1ZWNmODUwYTc3MjRhMjFmZWQyNmU3YjU1MWE4YTE0ODdjYmMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.VrOeu5REqx_irnQkF34tXb3v2F5VxwBPovtf2J5HIhE)
> 
> So took you advice and added the "-" to code and it appears to have worked. No idea what the anomaly is with that phase but for now its working fine. Thanks for your input, I doubt I'd have figured out the fix

I would also try instead of negative ‚-‚ switching the sides of the cable of clamps. That helped me. Before it was showing zero. Now it works correctly