danpeig/ESP32EnergyMonitor #14: Calibrating the ZMPT101B, sharp peaks and troughs

timomer, 2024-12-08:
Hello,

I have setup the hardware and uploaded the code for Calibrating the ZMPT101B, but my charts are appearing as follows:
[image]

This is very different from the example shown, where the peaks and troughs are more spread out. Before I proceed to the next step, can I confirm that what I am seeing is correct, or have I made a mistake somewhere? 

Note I have not connected any of the Current Transformers yet (3 phase)

mackloo, 2025-01-29:
I am trying this myself, mine did almost have the same look to it as in the readme but it was way faster and I did not really notice any difference when I turned the pot, I left it at good enough. 

Can't give you any advice though.
You should tripple check that you have connected the correct pins I guess or try another pin to see if the first one is faulty.

Have you gotten further ahead? Im trying to calibrate my CT clamps but they seem to pick up noise because i'm reading around 5 amps with no load and even without the probes connected.. Do you have similar issues? I've read that the ADC pins on ESP32 boards are noisy but this seems ridiculous..

I've tried two different boards, it has gotten better, now i'm down to 12v reading from 77v but CT clamps is still around 5A.

timomer, 2025-01-30:
Hi, thanks for your feedback. I needed to get my project live for the start of this year so ended up buying a shelly 3em so I could just get it live. 
I'm currently renovating my work space but once done will pick this up as I have another project I will use it for. Hope to do so in a month and will report back