danpeig/ESP32EnergyMonitor #16: The project is not longer usable with ESP-home because the custom component was removed

GerLech, 2025-03-02:
To solve the problem a real ESP-home component has to be created.

danpeig, 2025-03-02:
It used to be only YAML and plain C++, now it requires Python coding as well...

I see no purpose for ESPHome anymore, it is making things more complicated.

I will rewrite the entire code in pure C++ and interface with Home Assistant via the MQTT API. It is easier, faster and future proof.

carferrer, 2025-05-01:
Hello, i've try this and compile and work again. Only add those lines:

external_components:
  - source:
      type: git
      url: https://github.com/robertklep/esphome-custom-component
    components: [ custom, custom_component ]

danpeig, 2025-05-01:
Thanks! I will try it and make a donation to the project.

carferrer, 2025-05-01:
The developer of the project: https://github.com/robertklep

I have also try to use it in local an works:

external_components:
  - source:
        type: local
        path: my_components

Inside folder esphome/my_components copy the 2 folders os the project "custom" and "custom_component"

closed 2025-05-03 by danpeig